armel/uv-k5-chirp-driver #1: 2 lignes warrning en download

f1ixx, 2024-02-21:
Lorsque je lance un download radio to PC, j'avais 2 warrning sur 2 zones memoire :
~/source/radio/uv-k5/chirp/chirp/drivers/uvk5_egzumer_f4hwn.py:956: DeprecationWarning: Unnecessary #seekto 0xe80
  self._memobj = bitwise.parse(MEM_FORMAT, self._mmap)
~/source/radio/uv-k5/chirp/chirp/drivers/uvk5_egzumer_f4hwn.py:956: DeprecationWarning: Unnecessary #seekto 0x1FF2
  self._memobj = bitwise.parse(MEM_FORMAT, self._mmap)
Il semble que ce n'est pas genant lors du upload.

J'ai corrige avec :
#seekto 0xe80;  --> //#seekto 0xe80;
#seekto 0x1FF2 --> //#seekto 0x1FF2

Plus de ligne de warrning. Juste plus propre.
Merci pour vos travaux (firmeware et drivers)

Boris

note : teste uniquement sur Fedora 39 et Firmware v2.0

joc2, 2024-02-24:
ok, je ne voie pas de warning sous win 10 64 bit

Vous avez les warning dans le log de chirp ???

joc2, 2024-03-02:
effectivement c'est juste des warning.  Le but etait de trouver rapidement dans le code ou est defini tel option en espace memoire,  et oui elle sont consecutive, d'ou le warning.

pour le version francaise de chirp .... bonne idee. j'avais ca dans ma liste de chose a faire....
Si vous voulez la publier dans le repo de chirp que je fais sous armel, laisser moi savoir. je l'integerrais dans les futurs release de chirp. 
je vais vous laisser la version francaise. vu que vous l'avez deja debuter.
il y a tout les boites d'information qui s'affiche avec la souris qui faut mettre en francais aussi  (set_doc)

dans le code il y a une section help user qui est encore en commentaire #  pour l'instant
 des quelles sera completer, je la mettrerais active dans les futures release.
j'attend une reponse des programmeur de chirp pour pouvoir integer des fichier ( txt et image ) help directement dans chirp.
des fois juste du texte c'est pas evident pour exliquer quelque chose, une image vaut mille mot   :) 

[image]

73
:)

joc2, 2024-03-20:
le help file a ete publier

closed 2024-03-20 by joc2

f1ixx, 2024-03-20:
Bon[jour/soir]

wget fait ;-)
Je vais publie une version FR, mais doit revoir la detection de signature
car j'ai modifier pour ma version du firmware (compilation maison)

Merci pour le travaille fait

joc2, 2024-03-20:
ok. laisser moi savoirje veux l essayer avant de publier...joc2

Yahoo Courriel : Rechercher, organiser, conquérir

f1ixx, 2024-03-21:
Bonjour

J'ai mis le fichier la : https://github.com/f1ixx/uv-k5-chip-driver_F4HWN_FR/
Possible que je fasse encore des modifs. Mais du coup ca sera annoncer.

Merci
Bonne journee

Boris

joc2, 2024-03-21:
Bonjour Boris
Je viens d'essayer le fichier...
c'est pour votre version a vous car f4hwn n'a pas 144-146, 430-440, PMR  dans le flock ni d'option vox built remove etc....

j'ai quelque commentaire.
Vous avez enlever le nom egzumer du driver ... pourquoi ????
Il ne devrait avoir aucune reference sur aucun pays. Le logiciel est international. j'ai vu france dans un message...
Vous avez changer la facon de monter ce qui est vu sur le radio...normalement tout les menu vu sur le radio sont entre parenthese () a la fin de l'option, il ne sont plus entre parenthese et sont rendu au debut....
Tout les utilisateurs de egzumer et f4hwn sont habituer avec cette affichage entre parenthese...

Vous avez ajouter les numeros des menus entre parenthese.... hum...  Je voulais le faire moi aussi, mais on ne peux pas faire cela car selon la compilation des options dans le firmware, le numeros du menu va changer. donc ca ne fonctionnera pas selon si le firmware est compiler avec ou sans tel build option...
Il y a quelque erreur de traduction et d'ecriture.
c'est une bonne job que vous avez fait
mais pour l'integrer directement a f4hwn, je vais le modifier un peu. Avez vous objection. ??? Je vous y indiquerai comme traducteur du driver.
Jocelyn

f1ixx, 2024-03-22:
Bonjour

Pour Egzumer, oui j'ai hesite. Pour mon avis, vue le travaille de Armel, je pense plus parle
de la version F4HWN et de laisser dans les explications une info du fork. De plus, les indications
sont dans le firmware et sur les pages de doc.

Les nom afficher sur le talky me parle plus que de cherche la definition. Simple pour Mic, par exemple
mais Sauvegarde en memoire, pero, je prefaire avoir ChSave en debut de ligne. Mais comme deja dit
on met pour que tout le monde soi a l'aise dans l'utilisation. Idem les numero etait la pour m'aide
a retrouver sur la talky rapidement (saisie direct) pour aller changer les options. A supprimer.

Je vais commencer par reprendre tout ca pour mettre a jour sur mon github.
Pour les erreurs de traductions et autre, faudra passer dans un mon correcteur d'orthographe.
Ca n'est, comme vous l'avez vu, pas mon fort.

Pour la Bande 144-146 430-440 PMR, oui pardon, mon firmare est bricoler comme ca.
Mon talky a : zucune emission, uniquement PMR ou la bande de frequence ci dessus.
Je remets ca au propre.

Pour france, c'est dans la section Scramb.
Cette fonction permet de 'brouille' les commuications. Comme c'est la version en francais, j'annonce
que cette option est interdit en France. C'est identique dans d'autre pays.

Bref, je reprends tout ca et pousse dans github.

Nul besoin de mettre mon nom quelque part. 

Bonne soiree (/journee)

Boris

joc2, 2024-03-22:
je vais prendre votre traduction et l'adapter a f4hwn et vous l'envoyer.
vous me direz ca que vous en pensez.

f1ixx, 2024-03-23:
Pas de soucis.
L'essentiel est que, peu etre,  je vous ai fait gagne un peu de temps et pas trop en perdre.
J'ai fait des modifs avec les remarques du dessus.
73's

joc2, 2024-03-23:
Je viens de publier un release driver 2.3.0 fait a partir de votre version traduit du driver . Merci pour l'aide
J'ai fait la version francaise du document d'aide qui est sur github.
 
C'est tout inclu dans le release

f1ixx, 2024-03-23:
Je viens de voir la publication :+1: 
Merci a vous

armel, 2024-03-23:
Merci pour ton aide Boris ;)

Armel F4HWN.

f1ixx, 2024-03-27:
Bjr

Pas grand chose par rapport a votre travaille et celui de Jos2.
Merci

73's

Boris