armel/uv-k5-chirp-driver #3: S-meter - Codeplug reporting through Chirp (Advanced Settings)

F6EGK, 2024-03-07:
Bonjour Armel,

Ma question porte plutôt sur une clarification, qu'un réel bug. J'ai bien relevé et vérifié que le niveau à S9 correspondait à -93 dBm sur l'échelle du S-mètre. Cependant quand on rapatrie le codeplug via Chirp et votre driver associé, c'est une valeur à -76 dBm qui est affichée. 

[image]

J'ai l'impression qu'il s'agit d'une ancienne valeur correspondant à l'échelle adoptée en dessous de 30 MHz (-73 dBm), avec un point de basculement S à -76 dBm. Je me suis bien gardé de modifier quoi que ce soit dans ces champs. Je ne suis d'ailleurs pas certain que ce soit modifiable par l'utilisateur. 

Et merci encore pour vos efforts à optimiser le code, avec si peu de réserve de mémoire. J'espère que votre démarche vers Quansheng portera ses fruits, ce serait effectivement gagnant-gagnant. Je ne sais pas si l'option de grapiller des octets au travers de la suppression de la fonction VOX a suscité des réactions, mais je préfèrerai cette option en première approche, que celle de supprimer la fonction de réception FM broadcast !

73 - Jean-Roger

armel, 2024-03-07:
Bonjour Jean-Roger,

Cette valeur est modifiable. Mais elle sera sans effet car je n'en tiens pas compte. Je pense que c'est un héritage du firmware d'origine.

En pratique, je t'explique comment est évalué le niveau S. La documentation du BK4819 fournie les informations nécessaires pour récupérer le RSSI (Received Signal Strength Indicator). Elle indique également comment convertir ce RSSI en dBm. Et ceci est fait par la formule de conversion suivante:

`dBm = (rssi / 2) - 160;
`

Ca peut sembler un peu magique, mais c'est dans la documentation. Et c'est aussi vérifiable avec des outils de mesures, ce que j'ai fait, bien évidement. 

Pour ce qui est du Vox, le gain serait faible comparativement à la Broadcast FM :

Vox ~ 760 octets
Broadcast FM ~ 4 104 octets

73'

Armel F4HWN.

F6EGK, 2024-03-08:
Merci Armel pour toutes ces explications, notamment pour la conversion niveau RSSI vers dBm. Je ne vais pas vous embêter avec cela, mais du coup je suppose que ce qui figure sous le menu Calibration / RSSI correspond à des valeurs historiques du S-Mètre dans la version de firmware original. Rien à modifier ici aussi, ce serait donc une simple information.

Bien noté pour les perspectives d'espace mémoire récupérable, suivant les options possibles !

73 - Jean-Roger

armel, 2024-03-08:
Bonjour Jean-Roger,

C'est exactement cela.

73'

Armel F4HWN.

closed 2024-03-13 by armel