armel/uv-k5-chirp-driver #8: raccourci main only

stephane-rib, 2024-04-14:
bonjour Armel et tout d'abord merci beaucoup pour votre investissement dans le développement de ce firmware.

j'aurais une suggestion à propos du switch rx mode en raccourci, serait il possible de le modifier de façon à changer entre Main Only et le mode enregistré par défaut, cela éviterait de switcher plusieurs fois pour retrouver le mode que l'on utilise par défaut.
je sais c'est juste deux appuis supplémentaires mais ce serait plus ''ergonomique''

dans l'attente merci

Stéphane

armel, 2024-04-14:
Bonsoir Stéphane,

Attention, tu as posté cette issue sur le repository du driver Chirp.

Sinon, je comprends ta demande, surtout si tu es en général, en MAIN TX DUAL RX ou DUAL RX RESPOND.

Mais si tu es, par défaut, en MAIN ONLY (et pas mal d'utilisateurs le sont), l'utilisation du raccourci ne sera plus d'aucune utilité. Et si ponctuellement tu veux passer en DUAL RX RESPOND, par exemple, tu seras alors obligé de passer par le menu. Alors qu'avec la gestion du raccourci en vigueur actuellement, tu n'es pas obligé puisque ca cycle entre tous les modes.

Bref, tu vois, c'est plus compliqué. Pour bien faire, il faudrait pouvoir paramétrer 2 RxMode (un RxMode primaire et un RxMode secondaire) et le raccourci permettrait de passer de l'un à l'autre. J'y ai déjà pensé. Mais niveau espace mémoire, ce serait assez gourmand.

73'

Armel F4HWN.

stephane-rib, 2024-04-16:
merci pour la réponse,
désolé pour le mauvais endroit de dépôt
quant à ma demande effectivement c'est assez personnel et de plus gourmand en mémoire, je continuerais donc avec le switch entre les modes, après tout c'est assez rapide une fois que l'on a pris le coup et repéré l'icone d'état correspondant
encore merci
Stéphane

armel, 2024-04-16:
Bonjour Stéphane,

Oui, on s'y fait avec un peu d'entrainement. 
Je cloture cette issue, mais je garde tout de même ta demande en tête. Si je trouve un moyen simple et efficace pour répondre à ta demande, je le ferais.

73'

Armel F4HWN.

closed 2024-04-16 by armel