armel/uv-k5-chirp-driver #11: Soucis avec la programmation de touches F1 F2 M

14UVR10, 2024-06-14:
Hello et Bravo pour le boulot
Sur mon tx, j'ai trouvé un petit bugg
Lorsque je programme une des touches (F1 F2 M , même en long), la fonction  Bascule Wide/Narrox en direct sur le tx, tout fonctionne bien !
Dès que je charge le fichier sur chirp, cette fonction passe en mode PTT (CL ou OP).

Si je fais ma modif sur chirp et que je ré-injecte, Pareil
En gros chaque fois que j'utilise chirp, je dois retourner manuellement dans les menues pour valider la fonction W/N sur la touche programmée.
Salutation

armel, 2024-06-14:
Bonjour Alain et bonjour à Jocelyn,

Je confirme le bug remonté par Alain, avec le driver Chirp v2.8.
Je pense que cela vient de l'ajout de l'action MAIN ONLY (en plus de RX MODE qui existe toujours). 
Peut-être revoir, coté driver Chirp, la bonne prise en compte des actions.
Ensuite, aucune urgence, ca reste un hobby :)
Le reste semble fonctionner, d'après mes tests. Mais j'utilise principalement le driver Chirp pour enregistrer les canaux mémoire. 

73'

Armel F4HWN

armel, 2024-06-14:
Ok, je viens de jeter un oeil au code du driver.
Si ca peut aider @joc2, les actions disponibles sont, dans l'ordre :

- NONE
- FLASH LIGHT
- POWER
- MONITOR
- SCAN
- VOX
- 1750Hz
- LOCK KEYPAD
- VFO A VFO B
- VFO MEM
- MODE
- RX MODE
- MAIN ONLY
- PTT
- WIDE NARROW

Je pense qu'il y a quelques modifs à faire dans la déclaration de la liste 

```
KEYACTIONS_LIST = ["NONE",
                   "FLASHLIGHT",
                   "POWER",
                   "MONITOR",
                   "SCAN",
                   "Voice detection (VOX)",
                   "ALARM",
                   "FM RADIO",
                   "1750Hz TONE",
                   "LOCK KEYPAD",
                   "Main VFO (VFO A / VFO B)",
                   "Frequency/memory mode (VFO / MEM)",
                   "MODE",
                   "Put the backlight OFF temporarily (BL_MIN_TMP_OFF)",
                   "Change Rx Mode: (MAIN ONLY)",
                   "Toggle CLASSIC to ONE PUSH ptt (PTT)",                   
                   "WIDE / NARROW"
                  ]
```

Mais je préfère laisser la main à @joc2 sur cette partie ;) 

73'

Armel F4HWN.

joc2, 2024-06-14:
Bonjour,  J'ai pas ete voir le code de Armel... mon erreur , j'ai juste renomer les actions, dans la liste il y a tout les actions possible selon toute les option de compilation. les options non valide sont cacher dans chirp. c'est pour cela que la liste est longue dans le driver. example. blmin tmp off est dans la liste mais dans le code de armel elle n'est pas activer, donc chirp ne l'affiche pas.
on ne peut pas enlever un element car chaque element de la liste a un numeros ( position unique )  si on supprime un element toute les elements suivant l'element seront erroner...

je vais valider : Lorsque je programme une des touches (F1 F2 M , même en long), la fonction Bascule Wide/Narrox en direct sur le tx, tout fonctionne bien !   Dès que je charge le fichier sur chirp, cette fonction passe en mode PTT (CL ou OP).
 je vous reviens

[image].

joc2, 2024-06-14:
something not working has it supose to be.... i am looking about it

armel, 2024-06-14:
Note that I've added the MAIN ONLY action in v2.8, but that the RX MODE action is still available too. The MAIN ONLY action does not replace the RX MODE action :)

Armel F4HWN.

joc2, 2024-06-14:
that my mystake... i rename instead of adding main only... i am workink on

Yahoo Courriel : Rechercher, organiser, conquérir

armel, 2024-06-14:
No problem @joc2, this is the end of the week :)

joc2, 2024-06-15:
the change is done, i will publish soon... 2.8.1   
a weird thing happening 1 time....  the reading of the build option send by the radio was 0xff.... so it show all the option in the key programable option in chirp....
i close the radio and chirp, reload the driver open the radio, then, it's reading the radio built option normaly....
just continue make some test before send it to release...

closed 2024-06-15 by joc2