armel/uv-k5-chirp-driver #12: Beta Chirp - ScanList selection - Beta Chirp

KDMcMullan, 2024-07-05:
Really good work on V3. Thanks, as ever.

Ref v3.0 Beta Chirp Driver

For dropdown items, Chirp allows keyboard entry of items which begin with the keyboard item typed. (For example pressing "D" gets you "Disabled", pressing "E" gets you "Enabled".) This is a feature of Chirp and I don't imagine we can fix it.

The dropdown entries for ScanLists are None, List 1, list 2, etc. For rapid entry of scanlists, it would be very convenient if the entries started with the characters 1, 2, or 3 for list 1, 2 and 3.

Further, I propose that (again, for ease of keyboard entry):

- 1 & 2 might be "4: 1,2", 
- 1 & 5 might be "5: 1,3",
- 2 & 3 might be "6: 2,3", 
- 1, 2 & 3 might be "A: 1,2,3"

Each item now has a unique key: N, 1, 2, 3,4, 5, and 6.

Line 555:
`SCANLIST_LIST = ["None ", "1: L1", "2: L2", "4: L1 & L2", "3: L3", "5: L1 & L3", "6: L2 & L3", "A: All (1, 2, 3)"]
`

I don't see where the .PY is hosted, or I'd have a pull request on it.

joc2, 2024-07-14:
Wouaw,, i never use the keyboard to select a value.... i always use the mouse....  This is all over the code in every menu that can have this thing you just sugest to change.., never think about it...   i will look what involve to change all the menu.....

joc2, 2024-07-14:
Hi, i look about it and here what i see about this...
if i add this to all menu to be standart to all selection, this will add extra display data on each option, that come confusing a bit, since the number of the option is not usable for user.
other thing, in some menu, i remove item from the list depent on the selection, so the identification number will be mismatch. here is the try i do... so i came back the way is was...

[image]

closed 2024-07-14 by joc2

KDMcMullan, 2024-07-15:
Respectfully, I believe you have inadvertently closed the wrong ticket.

I'm puzzled by your two screenshots, which show nothing relevant to the suggestion. I do not with to set Timeout indicator to "Visual"

Can you review the decision to close this item, please, on the basis that mine was a simple, obvious, free, tiny modification? I have already made the 10 second modification in my own copy and it has saved me many hours.

Thanks.

KDMcMullan, 2024-07-15:
For clarity.
Mine was not a request to change the program; definitely not affecting "every menu this thing has".
This is a suggestion to change the newly added, not yet implemented, Beta version of the Chirp driver to make it more useable. Rather than having 3 clicks and navigation of a dropdown, I can select many many scanlists from the keyboard, without even thinking about it.
 
Hope this is useful.

joc2, 2024-07-15:
Hi.
the beta version is pass..
 the official  version has been release few days after Armel release the firmware 3.0
here is the link of the driver py.
Release Chirp driver v3.0.1 · armel/uv-k5-chirp-driver  
|  
|   
|   
|   |    |

   |

  |
|  
|   |  
Release Chirp driver v3.0.1 · armel/uv-k5-chirp-driver
 
error in the address for the scanlist priority and channel
  |   |

  |

  |

  
update it and this version fix the bug of the slist...

jocelyn

Yahoo Courriel : Rechercher, organiser, conquérir

joc2, 2024-07-16:
I understand, what you said. modify only the slist menu....
what i said, is to be the same process all over the option, uses the same way to select option all over the driver...
i dont know if just modify this option is a good idea....  i will need to ask Armel, what he think about it...
in the second screen shot, you see it!!!!, the strange display with the value selected.. that's why stop the modification.

KDMcMullan, 2024-07-16:
No. What you suggest would be quite a lot of work and there's little point, since most of the lists are uniquely keyed, or else lists of numbers. So, it actually mostly wouldn't be possible.
I propose modifying only the SList selection menu. It makes it a million times easier to program scanlists.
Also, please bear in mind: it's not REALLY really a modification: the driver is in Beta, and this extra option has only just been added.