armel/uv-k5-chirp-driver #13: AM Fix Option Still Present - Beta Chirp v3

KDMcMullan, 2024-07-05:
Pursuant to https://github.com/armel/uv-k5-firmware-custom/issues/165 (raised on the wrong repository), line 2282 of the Beta Chirp Driver v3 appears to allow the toggling of the AM fix.

joc2, 2024-07-14:
Yes, the amfix still in the driver...  this parameter has no effect , since F4HWN for it to be ON....   i will take a look to it i can force it to always ON... thank's for the info.

joc2, 2024-07-14:
Hi, i tough about this, i cant not remove it, if user use a version lower than 3.0 in their radio, this option must be accesible. i will put a note instead, some thing like this :  ( This option has no effect with firmware higher then 3.0 )

closed 2024-07-14 by joc2