armel/uv-k5-chirp-driver #14: Probleme avec les Mémoires avec chirp

Yves862, 2024-07-05:
J'ai utilisé la dernière version de chirp avec votre driver et ensuite j'ai mis le firmware f4hwn v3.0.
Le firmware est parfait, je l'adore.
Chirp vois bien le radio pour tout mais quand je transfert des frequences avec chirp rien ne reste dans la mémoire.
Mais à la main ça reste bien dans la mémoire.    Pouvez vous m'aider?

Yves862, 2024-07-05:
Ok merci
 Je vais attendre la version finale 
Bonne soirée à vous

armel, 2024-07-05:
Avec plaisir.
Bonne soirée à vous également.

Armel F4HWN

joc2, 2024-07-14:
Bonjour, Avec la derniere version avez vous toujours des problemes ??de mon cote tout fonctionnes

Yves862, 2024-07-14:
Je n'ai pas encore essayé.
Je travail trop, je manque de temps. Je suis en vacances des vendredi
prochain et je vais avoir du temps libre pour essayer.

Je vous tiens au courant .

Bonne soirée

Yves862, 2024-07-30:
Bonjour Armel

J'ai finalement eu le temps de m'amuser avec mon radio.
J'ai installé la dernière version (3.1) de votre logiciel avec la dernière
version de chirp et aussi la dernière version du driver pour chirp (3.1) et
malheureusement je n'ai pas retrouvé la capacité de voir les mémoires du
radio avec chirp.

J'ai présentement 69 poste enregistrés à la main dans le radio et quand je
téléverse du radio vers chirp, je vois les mémoires à l'écran mais elles
sont vident.
Donc c'est devenu un radio qui fonctionne pour les mémoires en mode manuel
seulement.

Je vais essayer de voir si quelque choses à été modifier dans les réglages
quelque part.

Si jamais vous mettez la doigt sur mon problème faite moi signe .

Merci à l'avance
Yves

armel, 2024-07-30:
Bonjour,

Je ne sais quoi vous dire.
Aucun problème similaire n'a été remonté. 
Et je ne comprends pas bien l'effet que vous observez : "je vois les mémoires à l'écran mais elles
sont vident" ?

Yves862, 2024-07-30:
> La version avec bandscope sans broadcast fm

J'ai effectué quelques test avec mon radio aujourd'hui et si j'installe IJV
3.20 comme sur mon autre radio,  tout les fonctions du radio fonctionnent
normalement.

Et quand je reflash pour votre version 3.1, chirp vois les mémoires vide
même si il y a quelques d'enregistrer dans la radio.

J'aime bien votre version alors je vais la laissé installer sur ce radio
même si je dois faire les programmations manuellement.

Merci pour votre bon travail

Tenez moi au courant si jamais vous trouvez quelque chose à ce sujet

Bonne journée

armel, 2024-07-30:
Houlala, ok.
Alors, je l'ai expliqué plusieurs fois et j'en ai même parlé dans mes vidéos.
Si vous installez IJV qui modifie en profondeur l'EEPROM, je ne réponds plus de rien.
Donc, ré-initialisé votre pocket, si possible avec une image de l'EEPROM AVANT l'installation de IJV.
Et tout rentrera dans l'ordre.

Armel F4HWN.

Yves862, 2024-07-31:
Oups!!!
C'est probablement une vidéo que j'ai manqué.

Merci pour la précision.

Je vais refaire mes devoirs
Et je vous tient au courant

Yves862, 2024-07-31:
Finalement tout fonctionne normalement maintenant.

Merci pour vos conseils .

Continuez votre bon travail
J'adore votre version d'edzumer.

Bonne journée à vous

closed 2024-07-31 by armel

londresarthur, 2024-08-14:
I'm having the same problem here

armel, 2024-08-14:
This issue has been solved, so please read all the messages above. 
The Chirp driver works very well, trust me.

londresarthur, 2024-08-14:
My problem was exactly the same as his, it's not your fault. To be more precise, the problem is how the IJV writes the memories as you use it.

For people who have this problem in the future: you need to do a RESET ALL inside the IJV, BEFORE installing any other firmware. If you are having this problem now, install the IJV again, do a RESET ALL, and then write the new firmware.

armel, 2024-08-14:
That's right @londresarthur and I agree with you. 

I've talked about it several times and even made a video on the subject. The IJV firmware radically changes the content of the eeprom (and the calibration). The best thing to do is always back up your eeprom BEFORE installing firmware (so directly out of the box). It is then possible to restore the factory settings afterwards.

londresarthur, 2024-08-14:
I had only backed up the calibration and configuration, but it wasn't enough. What software are you using to make the complete EEPROM dump?

armel, 2024-08-14:
I'm using [k5prog](https://github.com/sq5bpf/k5prog). 
This is a command-line tool.