armel/uv-k5-chirp-driver #21: Setting a scan list in Chirp with a single keypress

KDMcMullan, 2024-07-28:
It would be very convenient to be able to press a single key in Chirp to assign a particular channel to a scanlist. This allows rapid configuration of many scanlists from the keyboard, without clicking each (line item, and selecting the scanlist from the dropdown menu.

All that is required is that line 529 of the driver be modified as:

`SCANLIST_LIST = ["None ", "1 List [1]", "2 List [2]", "4 List [1, 2]", "3 List [3]", "5 List [1, 3]", "6 List [2, 3]", "All List [1, 2, 3]"]
`

This way, we can go through a list (potentially of 200 channels) without even looking at the mouse or the dropdown lists, and very easily assign each a scanlist number by repetitive keyboard strokes.

I humbly apologise if this is considered a duplicate, but its such an extremely obvious, completely free augmentation that I must have been misunderstood the first time I proposed it. I added this line to my ow version and was able to configure all myscan lists with the new system in a few minutes without giving myself RSI from 800 mouse clicks.

armel, 2024-07-28:
Hi,

I'll let Jocelyn answer that. 
But you know, it is possible to select several lines from Chirp. It's very fast. 

https://github.com/user-attachments/assets/b9bf04ca-33ec-4822-966d-c381f224dd12

Armel F4HWN

KDMcMullan, 2024-07-29:
That's a workable option, too. Thanks.

KDMcMullan, 2024-08-08:
I've changed my mind. It's a nightmare. It only works with contiguous lists of channels. If there's a gap, it just refuses to play.

armel, 2024-08-14:
So I'll close this issue.

closed 2024-08-14 by armel

KDMcMullan, 2024-08-15:
Ah, no.
I was saying that the selection of a list of channels and setting them all at once is a nightmare. It doesn't work if there's a gap in the list. I've implemented my fix in my local copy.

armel, 2024-08-15:
So remove the gap.

KDMcMullan, 2024-08-15:
I mean, if I want ch 1-10 to be slist 1 and ch 11 and 12 to be slist 2 and ch13-15 to be slist 1 and and ch16-20 to be slist 2, I cannot select ch1-10 and 13-15 all in one go to perform that one single operation. To remove the "gaps" I'd need to group together all the slist 1 channels, then all teh slist 2 channels, etc...

I'm really sorry. Mine is such a painfully obvious time-saver. I'm clearly not explaining it well.

armel, 2024-08-15:
@joc2 will get back to you (or not). My opinion is that you're making a big deal out of nothing. Seriously, this driver works like a charm even if I do most of the work directly from the transceiver.  
According to me, @joc2 has done a great job and that's good enough. 

Regards,

Armel F4HWN.

KDMcMullan, 2024-08-15:
He has. It does.