armel/uv-k5-chirp-driver #22: Puissance toujours en L2

fabjail, 2024-07-29:
F4hwn 3.1 bandscope
chirp 3.10.8
module chirp 3.0.1 

Bonjour,
la puissance reste toujours paramétrée sur L2 malgré la programmation chirp ou par le menu de l uvk5.
l uvk5 ne tiens pas compte de la programmation chirp "puissance" et en cas de modification manuel dans le menu uvk5, après extinction, la puissance repasse en L2.
j ai peut être raté quelque chose ,mais je ne vois pas.

en attendant , merci pour ce super boulot.

armel, 2024-07-29:
Bonjour,

Si vous utilisez le driver chirp 3.0.1, c'est normal.

Vous devriez essayer le [driver chirp 3.1.0](https://github.com/armel/uv-k5-chirp-driver/releases/tag/3.1.0) (le dernier en fait...). Ca devrait mieux se passer :)

Armel.

fabjail, 2024-07-29:
évidemment !
qu' est ce qu' on peu être aveugle parfois !

merci

closed 2024-07-29 by armel