armel/uv-k5-chirp-driver #24: 3_3_0 driver not loading

Linden1, 2024-08-21:
I'm running the latest chirp (CHIRP next-20240821) on Ubuntu 24.04. Also the new drive works fine under windows 11
Driver 3_1_0 loads without error, but the latest driver (3_3_0)fails with:
`ERROR: Failed to load module: invalid character '·' (U+00B7) (uvk5_egzumer_f4hwn_ver_3_3_0.py, line 112)
ERROR: <function ChirpMain.load_module at 0x7015b163a7a0> raised unexpected exception
Traceback (most recent call last):
  File "/home/gilly/.local/share/pipx/venvs/chirp/lib/python3.12/site-packages/chirp/wxui/main.py", line 1696, in load_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 991, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1129, in get_code
  File "<frozen importlib._bootstrap_external>", line 1059, in source_to_code
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/gilly/Documents/Radio/Drivers/uvk5_egzumer_f4hwn_ver_3_3_0.py", line 112
    <title>uv-k5-chirp-driver/uvk5_egzumer_f4hwn_ver_3_3_0.py at main · armel/uv-k5-chirp-driver</title>
                                                                      ^
SyntaxError: invalid character '·' (U+00B7)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gilly/.local/share/pipx/venvs/chirp/lib/python3.12/site-packages/chirp/wxui/common.py", line 643, in run_safe
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/gilly/.local/share/pipx/venvs/chirp/lib/python3.12/site-packages/chirp/wxui/main.py", line 1700, in load_module
    raise Exception(_('Invalid or unsupported module file'))
Exception: Invalid or unsupported module file`
3_1_0 driver is 134.6kB, 3_3_0 is 1.6MB(?). If you need any more just ask.

armel, 2024-08-21:
It's strange. Take a look to the size...
[image]

Linden1, 2024-08-21:
I knew the size was wrong, that's why I brought that up.
I wonder if the server had a hiccup. I've deleted and downloaded (for the third time) and all is fine now. Fixed......
but definitely strange.

Linden1, 2024-08-21:
Ooops! I downloaded the raw file from the code page.
Mea culpa!

armel, 2024-08-21:
I was thinking something like that.

closed 2024-08-21 by Linden1