armel/uv-k5-chirp-driver #26: Export des listes de Scan.

xavbo, 2024-10-04:
Bonjour et merci pour ce que vous faites.
Nous apprécions.

Lors de l'export d'une configuration de L'UK5, la colonne liste de Scan n'est pas exportée.
Du coup, a chaque fois, il faut les ressaisir à la main. 
Est ce possible de l'intégrer dans le fichier CSV de l'export.

Merci.
Xavier

armel, 2024-10-04:
Bonsoir Xavier,

Je pense que c'est une limitation de Chirp. Voir la [documentation officielle](https://chirpmyradio.com/projects/chirp/wiki/CSV_HowTo#Handling-missing-columns).

Bref, chirp se contente d'exporter, en standard et quelque soit la radio, les champs génériques suivants :

`Location,Name,Frequency,Duplex,Offset,Tone,rToneFreq,cToneFreq,DtcsCode,DtcsPolarity,RxDtcsCode,CrossMode,Mode,TStep,Skip,Power,Comment,URCALL,RPT1CALL,RPT2CALL,DVCODE
`

Je laisse néanmoins @joc2 nous préciser si cela peut-être modifié par le biais du Driver. Mais j'en doute. 

Armel F4HWN.

joc2, 2024-10-04:
Bonsoir Xavier,

Effectivement, Armel a bien repondu, c'est une limitation du logiciel chirp et non du driver pour Armel chirp que je fais...

je vais poser la question au team chirp pour voir si il y aurrais une possibiliter d'exporter plus d'info...

Jocelyn VE2 ZJM

joc2, 2024-10-05:
Bonjour a tous,

JE vien d'avoir la reponse est c'est negatif...

 la question:

Hi, is their a way to export all the data in the memory, include all the extra parameter.
i see https://chirpmyradio.com/projects/chirp/wiki/CSV_HowTo#Handling-missing-columns
but since then, maybe their is something to do with this.

la reponse 

Nope, CSV is the interchange format, so it only carries settings that are applicable to all models. You can import from any file type though.

closed 2024-10-05 by joc2

xavbo, 2024-10-05:
Bonsoir. Grand merci à vous deux pour votre retour et le temps passéXavier