armel/uv-k5-chirp-driver #27: Duplex Misalinement?

KDMcMullan, 2024-10-05:
If I set Duplex to "off" for a particular channel, when I upload to the radio, it sets -ve offset. In order to get no offset, I must leave the field blank.

I found the solution by accident, but it was a bit confusing having deliberately set duplex "off".

I have version 3.4 of F4HWN and of Chirp driver. I have a feeling this problem was present before v3.3.

Hope this helps.

joc2, 2024-10-07:
Tnank's for the info,, i will look about it, soon

joc2, 2024-10-13:
Hi, i found that their is more than off that are problematic, i work on it

joc2, 2024-10-14:
it's fix in version 3.5.1,
now if you select  off or empty the value it the same (off) 
i also now keep the value that is in offset instead of reset it to 0 when off of empty is selected, so if you want to go back with offset. the old value will be their.
when off or empty is select in dulex, the offset  is not display, ( this is a function of chirp main program, not the driver it self. )

[image]

[image]

so if you go back to duplex + or - , the offset value will be show again

[image]

closed 2024-10-14 by joc2

KDMcMullan, 2024-10-14:
How odd that no-one spotted it. Surely SOMEONE (even on some other firmware) at some point must have set this to "off" and wondered why it was still -ve!!

just6979, 2024-11-04:
This change broke the "fake tx disable by setting tx to 0 MHz" part. Now setting duplex to "off" in chirp does nothing, it acts the same as <empty>, which is **very bad** since this is the only way to prevent TX on a channel. (ever since the TX Lock function got changed to act _only_ as an allowlist for channels outside the F Plan). I can only hope I didn't accidentally TX on some channels that I intended to be RX-only before I noticed this and reverted the changes from 3.5.1.

Also, it seems that if I use the 3.5.0 duplex code, set duplex to off, _and_ set TX Lock to "ON", I actually get a normal, not faked, TX DISABLE, even for freqs inside the F Plan...

(Side note: what's up with this repo and putting the version in the filename and the whole archive folder? The repo history **is** an archive of previous version if anyone ever needed an older one. Also, with the version in the filename I have to update my chirp shortcut/alias everytime a new version of the driver comes out, which is super annoying.)

armel, 2024-11-04:
Hi,

If you want to memorize a frequency that is **inside your FLock** band plan, and blocking it from transmission, you can. Simply add a positive or negative offset with an out-of-band offset. For example, I'm using HAM CE band plan which is open between 144-146 MHz. If I add 145.250 MHz (inside band plan) and add a positive offset + 300 MHz --> TX DISABLE 

About versioning, you're probably used to using Git and GitHub. But you have to understand that 99.999% of the users of this Chirp driver and firmware don't know much about development and computing.

Armel.

just6979, 2024-11-04:
I don't want to have to manage offsets and duplex up or down manually. I just want to set Duplex = off to prevent TX like pretty much every other radio+firmware in Chirp. The same way this radio+firmware worked before 3.5.1...

To prevent completely jacking this thread, I did open https://github.com/armel/uv-k5-chirp-driver/issues/33 for further discussion of managing the Duplex settings.

As to the versioning, I didn't expect such a quick response, so I ended up opening https://github.com/armel/uv-k5-chirp-driver/issues/34 to further prevent jacking this thread.