armel/uv-k5-chirp-driver #28: user error or missing? BCL Busy Channel lockout menu 10/67 setting missing in chirp and 2 suggestions

sparky93a, 2024-10-14:
hi, as subject,

1. user error or missing? BCL Busy Channel lockout menu 10/67 setting missing in chirp, or im just being blind
2. suggestion - add menu numbers next to chirp settings as a reference.
3. suggestion - squ menu 57, i feel this should be menu 2 for quicker access.

thanks.

armel, 2024-10-14:
Hi,

As for BusyCL, take a look to this short video.

https://github.com/user-attachments/assets/47b1d9ec-0d81-4683-92a1-c0c0f4543f86

As for the menu number, there are too many changes in the firmware to update Chirp every time. You know, developing firmware (in C language) and maintaining a Chirp driver (in Python language) takes time. 

As with SetPwr (menu 57), this menu is located in the same place as all Set* menus. As I was talking to another user today, trust me, no order will make sense to everyone. 

Armel F4HWN

sparky93a, 2024-10-14:
thank you for your reply, hiding in plain sight!! and thanks you for your efforts all round here :-)

closed 2024-10-14 by sparky93a