armel/uv-k5-chirp-driver #30: Tx Lock doesn't seem to do anything

KDMcMullan, 2024-10-15:
Hi. What is Tx Lock (Txlock) for? I don't thin it's a native feature of Chirp: I think it's a feature of the driver. I was hoping it would disallow Tx on certain channels, but it doesn't seem to work for me. I have several channels programmed in (all within the hardware capability of my HT and within my licence) but setting or unsetting the lock does not disallow me from transmitting. Can you help, please? Thanks.

armel, 2024-10-15:
Hi,

Please, read the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus).

Armel F4HWN.

KDMcMullan, 2024-10-15:
Aaah. I had understood the complete converse: that items INSIDE the bandplan could be LOCKED.
In fact, I now understand the reality that items OUTSIDE the bandplan can be UNLOCKED.
This is tested and confirmed working as expected. Closing as a non-issue.
Thank you.

closed 2024-10-15 by KDMcMullan