armel/uv-k5-chirp-driver #33: Duplex change in 3.5.1 broke "duplex = off"

just6979, 2024-11-04:
"Duplex = off" does nothing since 3.5.1. See https://github.com/armel/uv-k5-chirp-driver/issues/27#issuecomment-2453652907

It seems that "Duplex = off" is not being changed into "Duplex = -" as it used to be.

I _can_ do an upload with "Duplex = off" in Chirp, however those channels will TX until I use the radio menus to change the offset to "-". Then the "fake tx disable" of forcing TX on 0.000 MHz works as expected, because the offset is still equal to the freq (in the radio, Chirp still shows the offset as empty), it's just that the "Duplex = off" after uploading with >= 3.5.1.

On a slightly seperate note:
When Duplex _is_ "-", and the offset freq = channel freq (in the radio, Chirp still shows it empty), _and_ TX Lock = ON, the the radio actually shows TX DISABLED as expected, instead of the weirdness of literally trying to TX on 0.000 MHz. I guess this makes sense, since the TX freq of 0.000 MHz is outside of any F Plan except Unlock All, thus allowing TX Lock to act as a denylist.

However, what would really be nice is if TX Lock simply acted as both an allowlist for freqs outside the F Plan as it does now, and also a denylist for freqs inside the F Plan (without needing the offset=freq weirdness). This way we could keep the real offsets stored for channels in the F Plan, but prevent TXing on them without first going into the menu and turning TX Lock to OFF. This can be useful if you want a duplex channel to act as RX-only, but also want to be able to sometimes listen in Reverse mode as well.

armel, 2024-11-04:
The real question is: why save a memory channel that is in my FLock band plan, but on which I don't want to transmit? Use case ? 
By the way, maybe your band plan isn't suited to your use? Also, if you're not a radio amateur, it's probably more efficient to use DISABLE ALL and take care of the TX Lock?
My 2 cents.

just6979, 2024-11-04:
I usually just use the Default F-Lock because it contains the US Amateur, MURS, and GMRS freqs, but also many Public Safety freqs that I used to easily disable TX with Duplex = Off.

Perhaps I will switch to FCC Ham F-Lock and use TX Lock = Off to enable the GMRS and MURS channels I want.

So yes, that is a valid way to acheive what I want, but it doesn't fix that "Duplex = Off" is broken and does not act as expected for a regular Chirp user. Perhaps that option should be removed from the driver if it's not fixed to fit with Chirp expectations: "If the duplex is set to off then transmission on this channel will be disabled (if the radio supports this capability)."

armel, 2024-11-04:
> I usually just use the Default F-Lock because it contains the US Amateur, MURS, and GMRS freqs, but also many Public Safety freqs that I used to easily disable TX with Duplex = Off.

Sorry but, using DISABLE ALL and using TXLock to open exceptions, is probably the best solution here, not the dirty Duplex OFF hack :)

> Perhaps I will switch to FCC Ham F-Lock and use TX Lock = Off to enable the GMRS and MURS channels I want.

Very good idea, I can only encourage you to do it. How many GMRS and MURS channels ? Less than 20 ? It's easy and quickly to do. 

> Perhaps that option should be removed from the driver if it's not fixed to fit with Chirp expectations: "If the duplex is set to off then transmission on this channel will be disabled (if the radio supports this capability)."

The important words here are "_if the radio supports this capability_" :)

In my opinion, Duplex can be + or - AND with an offset, or OFF for a simplex frequency. That's all and that's how it works today. Then, you can also play with setting an offset outside the band plan, but that's dirty. 

I'll let @Joc2 give his opinion. But for me, how it works today is nice. 

Armel.

just6979, 2024-11-04:
Yeah, it's only about a 20% of my memory channels that will need an override, so I'll go with FCC Ham FLock and open them with TX Lock Off. Guess this confusion is what I get for adapting my configuration solely based on other less flexible (no FLock plans, no TX Lock options except Duplex Off) radios!

"The important words here are "if the radio supports this capability" :)"

Ah, but the radio/firmware, doesn't it? The concept and functionality of "TX Disable" is present in this firmware, it's just only used for F Lock funtionality. I'm pretty sure "Off" won't be an option if the radio doesn't support using it for TX Disable.

"In my opinion, Duplex can be + or - AND with an offset, or OFF for a simplex frequency"

Except Chirp says "Note that for simplex channels, this should be set to (None)".  In fact, it looks like setting None in the 3.5.2 driver forces it to Off, which against the Chirp recommendation and unlike other radios I've used with Chirp. Perhaps Off could force it to None for clarity, or maybe better that Off should be removed if it's not actually supported. I guess we'll let @joc2 decide.

Thanks for the quick and thorough responses! Always interesting to see how different folks think about both bypassing and enforcing limits and configurations and such.

joc2, 2024-11-07:
Hi , i just see that, 
where is this :  Except Chirp says "Note that for simplex channels, this should be set to (None)". 

i dont see where you see it.

just6979, 2024-11-07:
https://chirpmyradio.com/projects/chirp/wiki/MemoryEditorColumns#Duplex

Specifically says that simplex channels should be set to None and RX-only should use "Off". Using "Off" for simplex is unusual and doesn't match Chirp's own recommendations.

With TX Lock support, the "Off" duplex mode should be removed. This is actually better than the hack of "Off" for RX-only, since it keeps the offsets in radio memory so Reverse/Talkaround can be used. for listening on the offset freq. But "Off" for simplex along with the forcing None to "Off" is confusing.

armel, 2024-11-07:
@just6979 Ok Justin, sounds good to me. So the best solution is probably to delete “Off” and leave only “None”, “+” and “-”.
@joc2 your advise ? You're the key master of this Chirp driver 😉

joc2, 2024-11-07:
Hi, this is the first time i saw that page , i dont know that chirp describe functionality of those setting, Chirp is the base software we uses for many radio and the ( driver i program )  is adapt to the radio specification...  not the reverse normaly.... but your right in this case, 

But more confusing. chirp do not allow None has a option of selection, ( this is built in chirp main program) 
even if they say in that documentation web page that None can be set...
here is the test i did to see what chirp list is allow. and it pop me this error.

debug log of chirp...
[2024-11-07 03:14:52,396] chirp.wxui.memedit - ERROR: Failed to load memory 138 as error because: `None' is not in valid list: ['', '+', '-', 'split', 'off']

So, i can't put None in the list, the only thing i can put is empty and remove the display of the off. Empty mean NONE. 

[image]

what do you think in that case

armel, 2024-11-07:
@joc2 Ok for me. Delete “Off” and leave only “+”, “-” and "".

just6979, 2024-11-07:
Sorry, I guess I didn't ever specify that I mean the empty string "" for None. I used "" sometimes, but then saw that Chirp docs usually uses None when the actual value is "". 👍

joc2, 2024-11-08:
ok, it will be done in the next release 3.6.2... maybe this weekend

joc2, 2024-11-10:
it's update on version 3.6.3

closed 2024-11-10 by joc2

baffles, 2024-11-29:
I feel like I'm missing something here. Is there a way to set tx lock for certain presets in chirp for rx-only? Do I really have to go through them one by one on the radio and save them back?

just6979, 2024-11-30:
You need to turn on "Show extra fields" in Chirp's View menu. Or pull up the properties for a channel or selection of channels and go to the Extra tab.

baffles, 2024-12-05:
Ah perfect. Thank you @just6979!