armel/uv-k5-chirp-driver #34: Why put the version in the file name?

just6979, 2024-11-04:
Putting the version into the filename is very weird and unusual, and it makes it difficult to try multiple versions if using an alias or shortcut to start Chirp with the driver loaded. I'd rather use Git to checkout the version I want to try, and easily switch back and forth with git, than have to change the way I launch Chirp each time I try a new driver version.

It also makes it harder to compare changes quickly, since the repo history is just constant new files and moved files. Instead of using Git to do the work of making diffs, you have to manually diff against the latest driver and various copies in the archive folder. This makes it difficult to bisect changes to find bugs, such as https://github.com/armel/uv-k5-chirp-driver/issues/33

I think you should consider going back to the pre-2.5.0 set-up, take the version out of the driver filename, and just rely on Git to keep the history for you. Perhaps keep the archive folder around for a while for reference. Or it could be converted into a series of commits to move the recent history out of the archive folder and back into the repo as is should be.

just6979, 2024-11-04:
re: https://github.com/armel/uv-k5-chirp-driver/issues/27#issuecomment-2453672314

Are those less technical users commonly needing to use old versions from the archive folder? They seem to know enough to get a brand new driver into Chirp for each update, shouldn't be too difficult a concept to, say get 3.5.0 by downloading from https://github.com/armel/uv-k5-chirp-driver/blob/3.5.0/uvk5_egzumer_f4hwn.py instead of searching through the archive folder for uvk5_egzumer_f4hwn_ver_3_5_0.py

just6979, 2024-11-04:
However, if you declare This Is The Way, I'm not going to push back any more than this. It's your repo, do as you like, I can work around it if I need to inspect any more changes, but I may also search around for other firmwares with the features I want, basically just: multiple scan lists, fast scanning, and a couple of the UI tweaks (SetGUI Tiny is super nice for DWR mode), and an easy way to make a channel RX only (and hopefully an active repo, which you should be commended for).

armel, 2024-11-04:
As I said about versioning, you're probably used to using Git and GitHub. But you have to understand that 99.999% of the users of this Chirp driver and firmware don't know much about development and computing. 

To give you an idea, every week I receive messages from people who have downloaded the driver web page (html format) and not the (raw) python file. Believe me, it's not easy for many of them. 

After that, if I'm the owner of this repo, it's @joc2 who does most of the work on the driver. And the whole community can thank him for it. 

Armel.