armel/uv-k5-chirp-driver #37: Paramètre F4HWN

J34NMY, 2024-11-09:
Hi,
Unless I'm mistaken on version uvk5_egzumer_f4hwn_ver_3_6_1_fr in the F4HWN settings we don't see the checkbox, to send only the F4HWN settings, nor the Set Timer setting.

armel, 2024-11-09:
Hi,

Thank you for your issue.
Since [Chirp driver v3.5.2](https://github.com/armel/uv-k5-chirp-driver/releases/tag/3.5.2), the upload of the FAHWN settings are now send with the standard setting. It's more simple. About SetOff setting, you're right. I fix it in the last [Chirp driver v3.6.2](https://github.com/armel/uv-k5-chirp-driver/releases/tag/3.6.2). Please, test it and tell me.

Armel F4HWN.

J34NMY, 2024-11-10:
Hi,
Thanks for the reply, indeed the modification on version V3.5.2 had escaped me. I have just tested version V3.6.2 the SetOff parameter and functional.
Best regards

armel, 2024-11-10:
Ok, thank you. Issue closed.

closed 2024-11-10 by armel