armel/uv-k5-chirp-driver #45: Chirp and UV-K5 menu item value discrepancy after reset

mattjm, 2024-11-27:
I am seeing some odd behavior in version 3.7.1 of the driver and 3.7 of the firmware.  Not sure if this is firmware or driver issue but we'll start here.  Feels like driver.  

If I "reset all" from the secret menu and then download the radio configuration in chirp, I get the following discrepancies between the radio menu configuration and what chirp shows:

| Menu Item | radio Value | Chirp Value | 
| --------- | -------- | ----------- |
|RX MODE|DUAL RX RESPOND|MAIN TX DUAL RX|
|SetOff|1h|OFF|
|KeyLck|off|15m|
|BatSav|1:4|1:5|
|TxTOut|1 min|N/U|
|BLTime|1 min| off |
|POnMsg|VOLTAGE|ALL|

The really interesting thing is then if I make _no changes_ and upload the configuration, the values above from chirp _do not_ overwrite the values on the radio.  However, if I change the value in chirp to something else and then upload it to the radio, the value is properly set.  

I tried this with two different radios and cables and got the same results.

armel, 2024-11-27:
First of all, Hi...

Okay, that may sound strange, but it's normal. When you do a Reset All, the eeprom is (almost) completely erased and the transceiver restarts. It will reread the parameters from the eeprom and, as they are empty, it will set default values (those you see from the transceiver, for example, SetOff to 01h00). 

But these values are only stored, for the moment, in memory (the microcontroller's memory). There has been no writing to the eeprom for the moment. The eeprom is therefore still empty. 

However, if you run Chirp, Chirp will read the contents of the eeprom and not the contents of the microcontroller's memory. Hence the difference in value.

Armel F4HWN.

armel, 2024-11-27:
Thank you Matt 😌