armel/uv-k5-chirp-driver #49: Skipping Channels During Scan

1jnewcs, 2024-12-05:
In chirp; the option to 'skip', gives no options. Normally there is an option 'S' to skip said channel during scanning. 

Also, setting some channels to: 'tx lockout': on, Doesn't lock them (doesn't show lock icon)

3.8.0

[image]

thx!!

armel, 2024-12-05:
First of all, good morning—it's free to say it.

Secondly, the Skip column **is not added by the driver** specific to my firmware, but by [Chirp itself](https://chirpmyradio.com/projects/chirp/wiki/MemoryEditorColumns#Understanding-CHIRPs-columns). This is a **standard column**. And I don't think it's possible to remove it. 

Thirdly, I believe you're referring to the TXLock column (and not TXLockout). I suggest you read the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus) if needed to understand how it works. Otherwise, please provide a concrete example of the issue.

Armel F4HWN.

armel, 2024-12-06:
Sounds good to me. You understand perfectly. 
Thank you.

closed 2024-12-06 by armel

1jnewcs, 2024-12-06:
Thank you and thanks so much for your work