armel/uv-k5-chirp-driver #55: How to disable TX on saved channel?

tomaz1, 2025-01-27:
I remember that in the Duplex column, it was previously possible to disable the TX of a channel. This functionality was removed, and saved channels with TX disabled now have Duplex set to "-" and the Offset set to the same value as the Frequency.

Currently, if I try to disable TX in the same way, I set Duplex to "-" and enter the Frequency value in the Offset column. However, I encounter an error:
Window Title: Invalid Entry
Body: Invalid edit: Tx freq 0.000000 is out of the supported range.

How do you correctly configure a channel to have TX disabled?

armel, 2025-01-27:
> I remember that in the Duplex column, it was previously possible to disable the TX of a channel.

Sorry but, according to me, it wasn't possible to disable the TX of a channel using Duplex column. 
Please, take a look to these closed issues #27 and #33.

Armel, F4HWN

tomaz1, 2025-01-27:
I have a line like this:

[image]

When I press PTT, the TX is disabled, which is exactly how I want it to be. I don’t want to accidentally transmit on this channel:

[image]

Now, I can’t change anything in this line/channel. For example, when I try to change the name of the channel, I get an error:

[image]

So, it’s still possible to have a "TX disabled" channel, but after one upgrade, this functionality stopped working. :(

What’s the new way to set a channel to 'TX disabled'?
Thanks.

FW on device is version 3.9.

armel, 2025-01-27:
I'm sorry, but I believe there’s a lot of confusion here.

- Your memory channel 200 uses AM modulation. It is **not possible to transmit** with AM modulation. You can only receive, not transmit. So, it is normal to see "TX DISABLE" on the screen.

- Your attempt to apply a negative OFFSET to the frequency 119.550 MHz is interesting. However, you are applying an OFFSET of 119.550 MHz. The resulting frequency will therefore be 0.000 MHz. This doesn’t make sense. The **lowest frequency supported by your K5 is 18.000 MHz**. Hence the error message you see with Chirp. Try 101.550000 as OFFSET, it will work... Try 101.650000 as OFFSET, it won't work (because 119.500 - 101.650 = 17.850 which is less than 18.000). 
 
In short, **everything is working as expected**. 

Now, here are a few questions I have:

- Are you a licensed amateur radio operator?
- What F Lock plan are you using?
- If you are just an SWL, use the F Lock DISABLE ALL band plan and then use the TX Lock menu to open a few transmit frequencies if needed (for example, your 16 PMR channels).

That’s the correct approach. About F Lock and TX Lock, everything is explained in the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus). Please read it.

Armel, F4HWN.

tomaz1, 2025-01-28:
Apologies, I gave a poor example earlier. However, I strongly disagree with your statement that Duplex=off does not work.

I searched for my old .img file from last April, opened it using ver_3_4_0.py, and confirmed that the line was configured like this:

[image]

 When I opened the exact same file with ver_3_6_2.py, it automatically changed to Duplex "-" and Offset "173.512500". I want to emphasize that I **did not** make these changes myself - this happened simply by opening the file in the newer version:

[image]

As an attachment, I’m including a screenshot that demonstrates the setting still works and TX is indeed DISABLED for this channel:

[image]

If you don’t believe me, I can send you the .img file to prove that the Duplex=off setting works on firmware version 3.9. The issue is that, in the current version 3.6.3 for Chirp, I can no longer edit this channel, for example channel name or list number. :(

armel, 2025-01-28:
I believe you, but again, please, take a look (an read the complete threads) to these closed issues https://github.com/armel/uv-k5-chirp-driver/issues/27 and https://github.com/armel/uv-k5-chirp-driver/issues/33 to understand why OFF option is not available anymore.

Armel, F4HWN

tomaz1, 2025-01-29:
Thanks, Armel.

If I still need to occasionally set a channel as TX Disabled (Rx only), I can use the older Chirp module, such as uvk5_egzumer_f4hwn_ver_3_4_0.py, where the Duplex=off option is still available and works for me.

Alternatively, if I understand correctly, I can achieve this by setting the Offset to a locked TX frequency range (F Lock) and configuring a large Offset value to match it. In this case, I also need to ensure that TxLock=ON is enabled (View->"Show Extra Fields").

This is just an example to test whether "TX DISABLED" is working as expected:

[image]

[image]

This solution works for me. Thanks.

closed 2025-01-29 by tomaz1

armel, 2025-01-29:
Hi !

Yes, that's right. 

Armel, F4HWN