armel/uv-k5-chirp-driver #57: Option for 4000mAh battery.

dineshalphaque, 2025-03-01:
The Ksun UV60D shares the same firmware and hardware specifications as the UV-K5. F4HWN Chirp drivers and firmware work perfectly fine on it.

The only difference is that it comes with a 4000mAh Li-Ion battery.

Wondering if this option and battery calculations for the battery meter can be included in? 

[image]

[image]

[image]

armel, 2025-03-01:
Hello,

This is not an issue related to the Chirp driver.

It's not enough to simply add a new option to the BatTyp menu. It's a bit more complicated 🤭 You also need calibration data. For example, for the 1600 mAh, 2200 mAh and 3600 mAh batteries, here are the corresponding data, directly include in the firmware code. They allow for a voltage-to-percentage mapping:

```
const uint16_t Voltage2PercentageTable[][7][3] = {
    [BATTERY_TYPE_1600_MAH] = {
        {828, 100},
        {814, 97 },
        {760, 25 },
        {729, 6  },
        {630, 0  },
        {0,   0  },
        {0,   0  },
    },

    [BATTERY_TYPE_2200_MAH] = {
        {832, 100},
        {813, 95 },
        {740, 60 },
        {707, 21 },
        {682, 5  },
        {630, 0  },
        {0,   0  },
    },

    [BATTERY_TYPE_3500_MAH] = {
        {837, 100},
        {826, 95 },
        {750, 50 },
        {700, 25 },
        {620, 5  },
        {600, 0  },
        {0,   0  },
    },
};
```

Without the data specific to the 4000 mAh battery, I can't do anything, sorry.

73'

Armel, F4HWN.

dineshalphaque, 2025-03-02:
Thanks for the explanation, @armel!

Would it help if I tried to take the voltage calculations at the percentage levels to help with these calculations in the firmware?

armel, 2025-03-02:
Hello,

> Would it help if I tried to take the voltage calculations at the percentage levels to help with these calculations in the firmware?

No. 

It's more complicated. You need to study the battery's discharge cycle. It's complex and time-consuming because it's not linear. You have to be extremely precise. This requires putting the radio in a stable state, without changing any parameters, and measuring the voltage evolution over time. With a 4000mAh battery, this will probably take 3 or 4 days, taking measurements every 30 minutes. You see the amount of work involved...?

Armel, F4HWN

armel, 2025-03-12:
Not closed by the author...

closed 2025-03-12 by armel