armel/uv-k5-chirp-driver #58: Permission to add latest Chrip module directly to Chirp

ijstokes, 2025-03-28:
I have acquired UV-K5s for many friends and family and flashed them with the excellent F4HWN firmware v4.0 fork/extension to Egzumer 0.22.  It is easy for me to do this step before I give them a flashed and configured UV-K5.  But it is much harder for me to get their laptops setup with Chirp that will allow them to program their F4HWN HT.

So I'd like to pick up from this attempt last year to do get a Chrip-ready version of the F4HWN driver in place:

https://chirpmyradio.com/issues/11270

But I'd like to know that I have the blessing of @armel before I try to take this on -- if I'm reading the Chrip issue comments from 11270 above correctly it seems like there are three main requirements:

1. Have a single internationalized driver, not an English and a French one
2. Make it clear how it fits into the UV-K5 ecosystem (is this a branch from the base UV-K5? or from Egzumer?)
3. Make sure it maximally sub-classes existing drivers (e.g. probably don't need to repeat all the standard CTSS and DCS codes?), in order to reduce the future maintenance burden.

TIA for any comments on how to advance this.  My approach would be to fork this repo, setup a PR, and then establish remote PRs back to this repo for review & inclusion, then only when @armel you were happy with the merged-to-this-repo version would I make an attempt to push a version to Chirp for their consideration.

armel, 2025-03-28:
Hi Ian,

We had already discussed this topic with @joc2, who has done a lot of work on this driver. It would be useful to get his input as well.

As for me, I have no experience with the process of integrating the driver directly into Chirp and no time to work on it. If you’re willing to handle it, I have no objections, of course.

A few things to keep in mind:

- My firmware may continue to evolve, even though I consider version 4.0 to be quite mature. This means the driver might need updates to follow these changes, and you’d have to reflect them in Chirp as they happen.

- While my firmware originally started as a fork of Egzumer’s, it has diverged significantly over time. Version 4.0 is quite different from Egzumer’s 0.22, even though they still share some common ground. I’m not sure if reusing existing subclasses for Egzumer would actually be beneficial in this case.

Thanks for your involvement in this project, and feel free to reach out if needed!

Armel, F4HWN

joc2, 2025-03-29:
Hi Ian,

Yes it's long text... but take the time to read it... 

I have done the chirp driver for egzumer since the begening, then we ( me and egzumer ) do the integration in chirp when egzumer firmware stop to have big modification ( become mature and egzumer think for him it's good enough ) , but it's was not a easy task the integration. They are many chirp rule need to be follow,  in the programming way (length code line has example) and in the thinking of show and display information in chirp for the radio, some are allow and some not....

With egzumer , chirp team help us to integrated, but that was not a easy task. a lot of time is consuming in this task...
but it finaly work :) , some change in firmware egzumer has been after the integration. so the driver need release in chirp team.

Then i start the driver for F4HWN and customise it a lot in the display, function, add many link web, etc.... ( 3 version of the firmware packaging )  

the problem with chirp integration instead of driver py file on the side, is when a change occur in the firmware of the radio, it need to be reflect in chirp in the same way in the same time. Many change become complex to handle change in chirp integration. but it's possible.

in the developpeent of the driver, i add the french driver version., since many user in french use it. and help develop firmware, chirp driver, bug , feature... pdf file, youtube... etc...

so to integrate in chirp , you will need a custom file for language traduction on each word sentence for chirp.  

a other thing, is the version driver with py file, we can load the appropriated version py depent of the firmware install in the radio.
in chirp this is only the active version that it's load. User need to be at the last version if they dont want corruption memory. 

if you want to look how integrate it in chirp,, it will be please to take a look with you. Since the firmware F4HWN is very good and way over the egzumer version. :)

If, we Start to do the integration, Armel will need to give us a bit of help, because we are not the ownwer of the program, Armel is...
 
The chirp team will talk with him at first to agree thing. Explain who we are.... might have some modification firmware in the radio to.
   
Their might be few thing that are not allow in chirp rules, that i do it in f4hwn driver...

when i was integrate thing in chip with egzumer,  the driver has to be re done to get in the chirp rules.

few time a go i submit a version of the F4HWN driver to chirp team , and the receive answers was, you need to make a sub class of the driver from the egzumer driver

But if you want to give help, let's go.

Everything is possible.

let me know.

Jocelyn

armel, 2025-03-29:
Hi Jocelyn,

Thanks for taking the time to explain everything in detail. I really appreciate your work on the CHIRP driver and all the efforts you've put into integrating it properly.

I fully agree that integrating the F4HWN firmware into CHIRP would be a great step forward. 

Beware however :

- if any firmware adjustments are required, this could be difficult, as there is no more memory space available (with the Bandscope edition),
- I probably won’t have much time to dedicate to it, but I’ll do my best. So the main work will rely on @ijstokes and you. Besides, as I already mentioned, I have no experience with the process of integrating the driver directly into Chirp.

Looking forward,

Armel, F4HWN.

ijstokes, 2025-04-04:
Thank you @armel and @joc2 for the detailed comments.  I am open to trying to make this work.  Also, I lived in Marseille and Nice alors je parle un peut de frrancais.  Ca marche bien pour les deux editions anglais/francais et le traduction.  I will attempt to investigate more this weekend.

armel, 2025-04-04:
Ok @ijstokes, sounds good to me. 

Just adding my two cents: if handling multiple languages makes development too messy, there’s no need to force it. I think everyone will be more than happy if we can finally drop the need for a custom driver. The real win is getting native Chirp support for my firmware. And let’s be honest — if the messages are in English, I’m pretty sure no one’s going to lose sleep over it. Now, if we can choose the language, of course, that would just be the cherry on top.

Merci à vous deux ;) 

Armel, F4HWN

RadioNostalgia, 2025-12-04:
Hi Everyone !!

my CHIRP refuse to load Python module!

HELP !!
'73 de IU2IEQ  Paolo Borroni